Tjoms99/xiao_sense_nrf52840_battery_lib #3: Killed my XIAO

schmidvon, 2024-02-27:
Hello, I don't know if this is an issue with your Lib but I added it to my project, connected a 3.7 Lipo. It first worked fine but after connecting and disconnecting an usb c cable to charge, my chip is now dead... could not determine what happened.

Tjoms99, 2024-02-28:
Hey @schmidvon, sounds very unfortunate. Could you tell me some more about your setup and code?

You say your chip is dead, is it the BQ charging chip or is the the nRF52 (microcontroller)? Is it only parts of the nRF52, like the ADC? Are you getting error messages in the log? Have you modified the library in any way? Have you tried reflashing the board? Maybe it would be a good idea to flash a simple Hello World or a blink example to make sure the problem is not code-related.

schmidvon, 2024-02-28:
Hello. I tested it with another Program as you said and I didn't kill the XIAO but the extension Board. I don't think this has anything to with your Lib. Thanks for your help!

closed 2024-02-28 by schmidvon