Tjoms99/xiao_sense_nrf52840_battery_lib #9: Optimizing power consumption

Tjoms99, 2024-06-23:
#### Summary
Addressing the issue of current draw by certain GPIOs when enabled, with a focus on optimizing power consumption.

#### Affected GPIOs
These three GPIOs are drawing current when enabled:

```
#define GPIO_BATTERY_CHARGE_SPEED 13
#define GPIO_BATTERY_CHARGING_ENABLE 17
#define GPIO_BATTERY_READ_ENABLE 14
```

#### Proposed Solution
When not in use, two of these GPIOs can be disabled to save power:

- `GPIO_BATTERY_CHARGE_SPEED`
- `GPIO_BATTERY_READ_ENABLE`

#### Details
- **GPIO_BATTERY_CHARGE_SPEED :**
  - Should be disabled when the charger is not connected.
  
- **GPIO_BATTERY_CHARGING_ENABLE:**
  - If disabled, the battery driver will not be able to detect when a charger is connected or disconnected.
  - Consequently, it will not be able to reconnect `GPIO_BATTERY_CHARGE_SPEED` automatically, and this would have to be done manually which is something we dont want.

- **GPIO_BATTERY_READ_ENABLE:**
  - Should be enabled just before a voltage sample is taken.
  - Should be disabled immediately after the sampling is done.

2opremio, 2024-09-30:
`GPIO_BATTERY_CHARGING_ENABLE` is particularly bad (it causes an extra current of 35uA). Could you please make it optional?

Also, are we supposed to copy your code into our projects to use your library? (that's what I did). It would be better to publish it as a [Zephyr module](https://docs.zephyrproject.org/latest/develop/modules.html)

2opremio, 2024-09-30:
BTW, huge thanks for publishing this!

Tjoms99, 2024-10-01:
Thanks for the feedback @2opremio! The idea is to automate the pin states as much as possible in the battery driver to lower the power consumption without so much thought from the programmer, keeping it simple. 

For more experienced programmers, enabling/disabling `GPIO_BATTERY_CHARGING_ENABLE` manually is definitely a feature that can be added. Consequently, the `GPIO_BATTERY_CHARGE_SPEED` would need to be manually enabled/disabled if the charging state is unknown. 

Cloning the repo and importing it to your project is the idea, but having it as a Zephyr module would be interesting; Thanks for the pointer! I will look into this when I have time. 

I see that you have forked the repo, are you planning on completing this issue?

2opremio, 2024-10-01:
> I see that you have forked the repo, are you planning on completing this issue?

For now I just stripped down the code to only do one-time measurements and enable the charge pin temporarily. I will take a look at contributing if I have time time.